inertialsense/inertial-sense-ros-old #3: Ref LLA Doesn't Seem to be working

superjax, 2018-03-05:
We need a method to easily input the reference LLA point, and have all estimates be with respect to that point (0,0,0) there

waltjohnson, 2018-03-05:
There is a refLla in the flash config (DID_FLASH_CONFIG.refLla[3].  However, that was intended to be static and not updated constantly.  Is this for GPS compassing?  We do have DID_GPS_RTK_MISC.baseLla[3] would could serve that purpose.

superjax, 2018-03-05:
No, this is about setting the static reference in the flash config.  I have had a few people ask me about this, or tell me it isn't working.  I think it's mainly a documentation issue.

waltjohnson, 2018-03-06:
It is set just like any other DID parameter.  DID_FLASH_CONFIG.refLla in decimal degrees. 

Maybe the location of refLla moved in the data set structure?

closed 2018-03-16 by superjax

superjax, 2018-03-16:
This was a bug that was fixed in `master`

waltjohnson, 2018-03-16:
Good to know!

*Walt Johnson *

*CEO & Founder* | *Inertial Sense, LLC*

Mobile: 801-380-0819 | InertialSense.com <http://www.inertialsense.com/>

[image: Worlds Smallest GPS-INS]

Electronic Privacy Notice. This message is confidential and may be
privileged. If you believe that this email has been sent to you in error,
please reply to the sender that you received the message in error and
delete this e-mail. Thank you.