inertialsense/inertial-sense-ros-old #10: Exposing the strobe to ROS

gellings, 2018-06-11:
I'm interested in having a camera provide a strobe to the inertial sense.  Ultimately I'm trying get the inertial sense time stamp of the camera images but I'm planning to start with the a navigation message at each image.

I've been looking through the sdk and the ROS code but I'm getting lost.  Could I get some help exposing the strobe in ROS?

superjax, 2018-06-11:
I can do this update, but I just need a pointer from @waltjohnson.  When do `DID_STROBE_IN_TIME` messages get published?  Is it just when the strobe in pin on the uINS goes from low to high?  Also, do I need to do any configuration of the communications to enable the strobe messages?

gellings, 2018-06-20:
I looks like this is working. Thanks for your help with this!

closed 2018-06-20 by gellings