inertialsense/inertial-sense-ros-old #12: Got GPS fix flag?

dallinbriggs, 2018-07-09:
Is there any flag that indicates the unit has received a GPS fix? In the old code there used to be a got_gps_fix_ variable that could be used to set the ref LLA from the first point the GPS was found. Is it just named differently now?

waltjohnson, 2018-07-09:
James will need to answer this question as it applies to
inertial_sense_ros.

However, in the InertialSense binary protocol
<http://docs.inertialsense.com/user-manual/com_protocol/binary/>, the INS
GPS fix status can be found in a couple messages:

INS_STATUS_NAV_FIX_STATUS(DID_INS_x.insStatus)

/** Navigation Fix Type */
enum eNavFixStatus
{
NAV_FIX_STATUS_NONE = (int)0x00000000,
NAV_FIX_STATUS_STANDARD = (int)0x00000001,
NAV_FIX_STATUS_RTK_SINGLE = (int)0x00000002,
NAV_FIX_STATUS_RTK_FLOAT = (int)0x00000003,
NAV_FIX_STATUS_RTK_FIX = (int)0x00000004,
};

or in DID_GPS_NAV.status

/** GPS Status */
enum eGpsStatus
{
/** Fix */
GPS_STATUS_FIX_NONE = (int)0x00000000,
GPS_STATUS_FIX_DEAD_RECKONING_ONLY = (int)0x00000100,
GPS_STATUS_FIX_2D = (int)0x00000200,
GPS_STATUS_FIX_3D = (int)0x00000300,
GPS_STATUS_FIX_GPS_PLUS_DEAD_RECK = (int)0x00000400,
GPS_STATUS_FIX_TIME_ONLY = (int)0x00000500,
GPS_STATUS_FIX_RESERVED1 = (int)0x00000600,
GPS_STATUS_FIX_RESERVED2 = (int)0x00000700,
GPS_STATUS_FIX_DGPS = (int)0x00000800,
GPS_STATUS_FIX_SBAS = (int)0x00000900,
GPS_STATUS_FIX_RTK_SINGLE = (int)0x00000A00,
GPS_STATUS_FIX_RTK_FLOAT = (int)0x00000B00,
GPS_STATUS_FIX_RTK_FIX = (int)0x00000C00,
GPS_STATUS_FIX_MASK = (int)0x0000FF00,
GPS_STATUS_FIX_BIT_OFFSET = (int)8,
};

*Walt Johnson *

*CTO & Founder* | *Inertial Sense, Inc.*

Mobile: 801-380-0819 | InertialSense.com <http://www.inertialsense.com/>

[image: Worlds Smallest GPS-INS]

Electronic Privacy Notice. This message is confidential and may be
privileged. If you believe that this email has been sent to you in error,
please reply to the sender that you received the message in error and
delete this e-mail. Thank you.

dallinbriggs, 2018-07-10:
Thanks Walt. I made a branch that sets the "home" position to wherever you start the ROS node and you have a GPS fix. This is similar behavior to several autopilots. It's on my fork on the release branch: https://github.com/dallinbriggs/inertial_sense.git

closed 2018-07-10 by dallinbriggs