inertialsense/inertial-sense-ros-old #13: /ins covariance indefinite

gellings, 2018-07-10:
I've noticed that the covariance in the /ins topic is not positive definite.  Specifically, the linear velocity portion of the covariance has negative Eigen values.  Just an FYI.  

This might be a real easy fix or it might show some potentially significant underlying problems.  I hope its the former.

waltjohnson, 2018-07-10:
Hi Gary.  You are correct.  It should always be positive definite.  Do you
know how we can recreate the scenario?  Are there certain things that cause
this issues?  Is it in NAV mode or AHRS mode?

*Walt Johnson *

*CTO & Founder* | *Inertial Sense, Inc.*

Mobile: 801-380-0819 | InertialSense.com <http://www.inertialsense.com/>

[image: Worlds Smallest GPS-INS]

Electronic Privacy Notice. This message is confidential and may be
privileged. If you believe that this email has been sent to you in error,
please reply to the sender that you received the message in error and
delete this e-mail. Thank you.

gellings, 2018-07-10:
Sorry, I assume it is in NAV mode.  I do have a GPS attached but beyond that the ROS interface is obscuring those details for me.  I don't have any other details about the scenario other than it often goes indefinite.

I think the problem might be [here](https://github.com/inertialsense/inertial_sense_ros/blob/7f5603c13f99df587e7bd309d52f794323830c17/src/inertial_sense.cpp#L225).  @superjax will have to confirm but it looks like the covariance was being rotated to be in the body fixed frame.  I'm not sure if this is right because variance isn't a vector. Though an absolute value might be good enough to fix it (I'd have to stop and think about the math more...).

There is another potential problem because the ins is using NED coordinants (which I prefer, for the record) but ROS convention is NWU.  Maybe that's for another issue.

superjax, 2018-07-11:
Thanks @gellings.  It looks like I am rotating the main diagonal, but not the off-diagonal terms.  I'm on vacation right now, but that should be an easy fix once I get back.  Probably Monday

closed 2019-06-25 by siddirp