inertialsense/inertial-sense-ros-old #14: ERROR: Quaternion Not Properly Normalized and Topics not being published

Manbeast-95, 2018-07-19:
Hi

I am trying to launch the inertial_sense_node. It shows that it has established connection with the uINS sensor along with a warning "MSG to TF: Quaternion Not Properly Normalized". Also /ins topic is not being published at all. When I try to launch using the default launch file test.launch, it shows the same error along with the phrase "bad data" and streams data continuously on the screen.

Any help to resolve this is appreciated.

Thanks

superjax, 2018-07-19:
Do you know the software version of your uINS?

Manbeast-95, 2018-07-19:
No. It is not available on the sensor anywhere. I have not changed any firmware on the sensor. It is just like how it was bought. Do I need to upgrade the sensor firmware? If yes, how can I do it?

superjax, 2018-07-19:
Yeah, can you bring the firmware up to the latest?  (At this point, it is 1.6.2)

https://github.com/inertialsense/InertialSenseSDK/releases/tag/1.6.2

You can use the graphical utility if you have access to a windows computer, and use that to flash the firmware.

If you don't, you can use the command line utility.  Here are the commands I just performed on my machine to build and run the cltool (change the file paths as necessary)
``` bash
cd InertialSenseSDK-1.6.2/
cd CLTool/
mkdir build
cd build/
cmake ..
make -j4 -l4
./bin/cltool -c=/dev/ttyUSB0 -baud=921600 -b=/home/superjax/Downloads/IS_uINS-3_v1.6.6.0_b2781_2018-06-29_175015.hex 
./bin/cltool -c=/dev/ttyUSB0 -baud=921600 -msgINS2
```

superjax, 2018-07-19:
I also just pushed a `1.6.2` branch that I confirmed works with the latest firmware.

superjax, 2018-07-19:
> Thanks. I updated the firmware using CLTool just as you suggested and
> result was success.
> I just cloned that branch and tried to test it and now I am getting a new
> error:
>
> "[FATAL] [1532028328.247633576]: inertialsense: No response when
> requesting flash configuration from uINS on "/dev/ttyUSB0", at 3000000 baud"
>
> Can you please help me solve this?

Did you do a submodule update?

Manbeast-95, 2018-07-19:
Yes. Still the same error and no data is showing up on any topic.
But the sensor is successfully connected as it shows here. 
[ INFO] [1532029836.154221301]: Connecting to serial port "/dev/ttyUSB0", at 3000000 baud
[ INFO] [1532029836.155681791]: Connected to uINS on "/dev/ttyUSB0", at 3000000 baud
[FATAL] [1532029839.156206723]: inertialsense: No response when requesting flash configuration from uINS on "/dev/ttyUSB0", at 3000000 baud

Manbeast-95, 2018-07-19:
The color is green with red flashing intermittently.

superjax, 2018-07-20:
It appears that the serial protocol submodule is messed up in the ROS node.  I just update the `release` branch to be up to date with `1.6.2`.  Could you please try blasting away the repository and cloning it down fresh, with the current `release`?

superjax, 2018-07-20:
Also, the autobaud detection only runs the first time the uINS is connected to something.  So, if you've connected to the CLTool at `921600` you either have to reset the device (one of the buttons) or change the `baudrate` parameter of the ROS node to `921600`

waltjohnson, 2018-07-20:
If need, there is a flash config bit on the uINS that disables the autobaud feature.

Manbeast-95, 2018-07-20:
It is working. 
I reflashed the firmware using CLTool and re-cloned the release branch. it is working. It seems to work on 3000000 baudrate as well.

Thanks for your support. I will close this issue now. I have some new issues regarding INS data which I will post as a new question.

closed 2018-07-20 by Manbeast-95