inertialsense/inertial-sense-ros-old #15: Bias and drift in /ins data even if the uINS sensor is stationary.

Manbeast-95, 2018-07-23:
The message below is a /ins topic message obtained when the sensor is stationary.  

It is clear that the twist data has a massive bias. And this bias is being translated into drift in pose data especially orientation.

1. Is there a way to address and correct this imu-bias in the ROS node before fusing for INS data?
2. Why is the pose.position data showing such high large values? I have already set GPS_ref_lla parameter as the co-ordinates and altitude of the origin of my map frame. Is anything else needed to correct this?

Your support would be greatly supported.
Thanks

`  header: 
  seq: 47054
  stamp: 
    secs: 1532365776
    nsecs: 541112410
  frame_id: imu
child_frame_id: ''
pose: 
  pose: 
    position: 
      x: 4710791.5
      y: -7905469.0
      z: 11.4687309265
    orientation: 
      x: -0.000432016066043
      y: -0.00188187009189
      z: 0.896768271923
      w: 0.442496418953
  covariance: [0.001568859675899148, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0015688366256654263, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.004249213263392448, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.575621240590408e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.7515457102490473e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.8030594617357565e-07]
twist: 
  twist: 
    linear: 
      x: 0.58526968956
      y: -2.11744022369
      z: -0.0384144186974
    angular: 
      x: 0.0125841256231
      y: -0.00336732831784
      z: -0.00795264542103
  covariance: [0.0003823317431077872, 4.3471229831687764e-05, -1.6403492579558617e-06, 0.0, 0.0, 0.0, 4.347122983168774e-05, 0.00040451053425531465, -1.0564898410849513e-06, 0.0, 0.0, 0.0, -1.640349257955862e-06, -1.0564898410849513e-06, 0.00011824383437990175, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 8.942446783066771e-11, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 8.9422497184799e-11, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 8.942591805949363e-11] `

jonathantullis, 2018-07-23:
Hey there!

Thanks for leaving us with your feedback.  We want to make sure you get the support that you need.

Our engineers are viewing your feedback but don't fully understand your question.  Could you more clearly explain your issue - what it is and what it is affecting?  The more information you can provide about your application and exactly what you need to accomplish, the better we will be able to address your issue.

Thanks!

Manbeast-95, 2018-07-23:
Consider the /ins topic message that I have attached above. This is a message braodcasted by the /ins topic when the uINS sensor is stationary.
With reference to that, my main two questions are as follows:

1. Since, the sensor is stationary, we expect both linear and angular twist data to be very close to zero. But if you look at the linear twist values, you see that the x-component of the linear twist is showing up as 0.585m/s and the y-component is showing up as -2.11m/s when in fact, both of those should be zero/very small and close to zero. So clearly, we have a bias in the sensor data. It is affecting further experiments because when I use this INS data to fuse with an EKF, due to these wrong  values and no way to model this bias, the filter will converge to the right model. So, why is this bias appearing? Do you guys have an error model for the sensor so that we can tune these biases out?

2. Now, Consider the Pose part of the message. If you take a look at the pose.position values, x component is 4710791.5 and y component is 7905469.0. I have set the GPS reference position parameter (GPS_ref_lla) to be the [latitude, longitude, altitude] of the origin of my map frame. So, I expect the X and Y positions to be reported relative to the GPS co-ordinates that I have set in GPS_ref_lla. Instead how are these huge values showing up?  Should I include any other transformations to get a good global position estimate and not these random huge values that I do not know how to make sense of?

superjax, 2018-07-31:
Sorry about the problem! 

I have had a hard time replicating your problem reliably, but I think I may have fixed it (likely an issue with mis-matched serial protocol versions).

Would you please try current `master` with the latest firmware `1.6.2`?  (I have added a `firmware_update` service to the ROS node so it should be a little easier to update the firmware)  Be sure to update submodules with `git submodule update --init --recursive`

Manbeast-95, 2018-07-31:
Hi

Thanks for the commit. I switched to master branch, updated the submodules but the problem is still there. I was already on the latest firmware version but re-flashed 1.6.2 again just in case but it doesn't seem to help the actual issues in /gps and /ins topics.
 
Let me know if I have to do anything more to help solve this issue.

superjax, 2018-08-01:
Sorry about that.  I'm a bit mystified.  Could you please help me a bit in debugging this?

Do you have the CLtool built somewhere?  

(download the source code from the [1.6.2 SDK release](https://github.com/inertialsense/InertialSenseSDK/releases/tag/1.6.2))
```
cd InertialSenseSDK/cpp/SDK/CLTool
mkdir build
cd build
cmake ..
make
```

and see if you observe the same behavior on the CLtool output?

```
./bin/cltool -c=/dev/ttyUSB0 -baud=921600 -msgINS2
```

Manbeast-95, 2018-08-01:
'DID_DEV_INFO: SN30641, Fw 1.6.6.0 r2781, 2018-06-29 17:50:15, Proto 1.2.74.6
DID_INS_2: 324850.437s
        Qn2b         0.365,  -0.000,  -0.002,  -0.931
        (Euler)      0.22,   -0.12, -137.18
        UWV         -9.0,     5.7,    -0.2
        LLA        42.3938713,  -71.1434845,  -20.2 ellipsoid
        STATUS
                Satellite Rx 1     INS Fusing: GPS 0, Mag 1
                Mode: AHRS         Solution: AHRS GOOD
                Errors    Rx parse 2, temperature 0, self-test 0'

This is the Ins message when viewed using CLTool. I am not sure what values here indicate the velocity and position data. Can you please elaborate on that?

superjax, 2018-08-01:
Yeah,

Position is the LLA term, while velocity is the UVW term.

So, based on that output, it looks like you are receiving GPS, but the system is still in AHRS mode, or basically not fusing GPS.  That is unusual.

Could you please create a log file using the cltool?

``` bash
./bin/cltool -c=/dev/ttyUSB0 -baud=921600 -msgPPD -lon -lts=1
```

This should make an `IS_logs` folder with a subdirectory of the log timestamp.  Could you let it log for something like 5 minutes, and upload the zip file of the log folder?

commit 4c59b9bd (2018-07-31): I think I fixed #15
closed 2018-08-01 by Manbeast-95