inertialsense/inertial-sense-ros-old #18: `No response when requesting flash configuration from uINS`

chunhochow, 2018-09-18:
I'm on the release branch (commit 4e6ba67). I followed all the firmware update instructions using the SDK in the NOTICE section in README.md .

However, when I run `roslaunch inertial_sense test.launch`, I get:
```
[ INFO] [1537293320.942305390]: Connecting to serial port "/dev/ttyUSB1", at 3000000 baud
[ INFO] [1537293320.943372816]: Connected to uINS on "/dev/ttyUSB1", at 3000000 baud
[FATAL] [1537293323.943677890]: inertialsense: No response when requesting flash configuration from uINS on "/dev/ttyUSB1", at 3000000 baud
[ INFO] [1537293323.945463717]: navigation rate change from 256ms to 10ms, resetting uINS to make change
[FATAL] [1537293329.946605037]: inertialsense: No response when requesting flash configuration from uINS on "/dev/ttyUSB1", at 3000000 baud
[ERROR] [1537293329.946728749]: inertialsense: unable to change navigation rate from 256ms to 10ms
```

However, when I run the connection via the SDK directly (the last command in the README's NOTICe section), I do manage to get data from the inertial sense device:
```
DID_DEV_INFO: SN31455, Fw 1.6.6.0 r2781, 2018-06-29 17:50:15, Proto 1.2.74.6
DID_INS_2: 12.317s
	Qn2b	     0.437,   0.071,   0.067,   0.894
	(Euler)	    10.44,   -3.90,  127.59
	UWV	     0.0,     0.0,     0.0
	LLA	   37.8769304, -122.2522608,  214.6 ellipsoid
	STATUS
		Satellite Rx 1     INS Fusing: GPS 0, Mag 1
		Mode: AHRS         Solution: AHRS GOOD
		Errors    Rx parse 0, temperature 0, self-test 0
```

superjax, 2018-09-18:
Could you try resetting the uINS before connecting to the ROS node?  (the autobaud detection only works the first time it connects after a reboot).  If you're using a evaluation board, one of the buttons is the reset button.  Or, you could just plug and unplug it.

chunhochow, 2018-09-24:
Thanks, that worked. Might be good to add that to the README.

closed 2018-09-24 by chunhochow