inertialsense/inertial-sense-ros-old #21: ROS driver does not publish any messages on /gps and /gps/info topics in the compassing branch

arunanshu-kittyhawk, 2019-01-11:
The test config is using a dual GPS uINS with FW version 1.7.2
I see messages being published on all topics except the /gps and /gps/info
This is also true on master branch.
The stream is parameter is set to true for both and the relevant topics are being advertised.
With an older uINS I can see /gps messages but nothing on /gps/info

Could this be an incompatibility with the latest 1.7.2 firmware?

https://github.com/superjax/inertial_sense_ros/issues/10

superjax, 2019-01-19:
This was an issue with 1.7.2.  I just upgraded the node to 1.7.3 and just tested this on my desk.  It seems to be fixed now on both `master` and `compassing`

arunanshu-kittyhawk, 2019-01-23:
I'm still seeing the same issue on my system. I'm on the latest compassing branch with v1.7.3 firmware on the uINS. 
Also, is the compassing branch required still or has the compassing feature been merged into master now?

arunanshu-kittyhawk, 2019-01-23:
Has the most recent commit in the repository worked for you?

siddirp, 2019-01-24:
@arunanshu-kittyhawk The compassing branch is still required for any compassing applications. Our testing shows both branches functioning properly.

siddirp, 2019-09-17:
@arunanshu-kittyhawk Is this still an issue? If so please let me know what questions you have. If not, could you please close this issue?

closed 2020-05-14 by siddirp