inertialsense/inertial-sense-ros-old #22: INS ros node publishes NaN.

BillThePlatypus, 2019-03-12:
I did a GPS test outside for a few minutes. When I came in to work with the logged data, I noticed that after a 121 seconds, the position, orientation, and linear twist all start publishing nan. Angular twist still publishes good values.
A bag file, with the ins data as well as ROSFlight data, is [here](https://byu.box.com/s/xl5f6gwq7cfvs0tvd3jdwi5czovx64il).

BillThePlatypus, 2019-03-12:
I had the gps topic on at the time, and it continued to publish correctly.

siddirp, 2019-09-17:
@BillThePlatypus Is this still an issue for you? Let me know if you have more questions. If not, can I close this issue?

BillThePlatypus, 2019-09-17:
It is not currently an issue. It has been a while, so I can't remember details, but this may have been a firmware version issue.

closed 2019-09-17 by BillThePlatypus