inertialsense/inertial-sense-ros-old #24: /firmware_update with rosservice failed

chunhochow, 2019-03-19:
I tried following the instructions in the README to update the uINS firmware, but it returns:
```
success: False
message: ''
```

What I did:
I first `roslaunch inertial_sense test.launch` (I couldn't access /firmware_update otherwise)
Then I ran the `rosservice call /firmware_update /path/to/IS_uINS-3_v1.7.3<...>.hex` (I tried both v1.7.3 and v1.7.5

superjax, 2019-03-20:
Did the lights flash on the device?

chunhochow, 2019-03-20:
It's unclear, since it's the Rugged uIMU unit, which means it has no lights.

I tried doing the firmware_update with my EVB1 IMU unit (that we bought last year), but it crashes immediately when I run `roslaunch inertial_sense test.launch`. The error it gives is `Unable to open serial port "/dev/ttyUSB0", at 921600 baud` (even though the device is clearly connected as `/dev/ttyUSB0`. (Is there another way to do this firmware_update without running `roslaunch inertial_sense test.launch`?)

waltjohnson, 2019-03-21:
We have several options for updating the firmware. If you're using Linux
and you're unable to get the ROS node to work properly you can use the
CLTool  <https://docs.inertialsense.com/user-manual/software/cltool/#cltool>that
is found in our SDK. The SDK can be downloaded here
<https://github.com/inertialsense/InertialSenseSDK/releases>. Instruction
on how to compile and run the CLTool can be found in our online
documentation
<https://docs.inertialsense.com/user-manual/software/cltool/#compile-run-linuxmac>.
There are also directions in the document on how to update the firmware
<https://docs.inertialsense.com/user-manual/software/cltool/#update-the-firmware>
using
the CLTool. Let's get the CLTool running and try to communicate with your
device and then we can figure out what is going on with the ROS node.

Andrew

>

chunhochow, 2019-03-22:
### Older Rugged uIMU
Firmware update worked with CLTool . It still returns `success: False` with no message using the ROS `/firmware_update`.

### Newer EVB1 IMU unit
Cannot update to firmware v1.7.5 with CLTool (stuck at blue pulsing light). I tried both the uIMU and EVB .hex files. Errors as follows:
```
Updating application firmware using file at /home/nvidia/Downloads/IS_uINS-3_v1.7.5_b160_2019-02-25_122633.hex
Starting bootloader...
Bootloader upload: 96%     
Results:
/dev/ttyUSB0: Failed to change to page 8
Failure issuing select page command for upload

1 ports failed.
```
```
Updating application firmware using file at /home/nvidia/Downloads/IS_EVB-2_v1.7.5_b28_2019-02-25_122655.hex
Starting bootloader...
Bootloader upload: 4%     
Results:
/dev/ttyUSB0: Failed to write checksum 71 at offset 34999
Failed to upload 255 bytes to page 0 at offset 34999

1 ports failed.
```

siddirp, 2019-03-26:
@chunhochow What is the version of firmware on the Newer EVB1 unit? It looks like you may need to update the bootloader before updating the firmware.

chunhochow, 2019-03-26:
@siddirp Woops I mixed up the labelling of "Newer" and "Older". It's fixed now in the previous comment.

EVB1 firmware version: I'm not entirely sure, but I think it's v1.7.0

Is there a way to update the bootloader without using EvalTool which requires Windows? (We only have Mac and Linux machines.)

siddirp, 2019-03-26:
Yes. In our SDK there is an example project called ISBootloaderExample. That project actually is setup to update the firmware. On line 101 if you replace bootloadFileEx with bootloadUpdateBootloader and fill out the arguments appropriately. You can then update the bootloader.

chunhochow, 2019-04-02:
Can you write a separate script with the correct arguments filled in for updating the bootloader? I assume that this would probably be a relatively common issue that other people would run into too. Thanks.

siddirp, 2019-04-10:
@chunhochow I will look into adding another project to our SDK. Were you able to successfully update the bootloader?

chunhochow, 2019-04-18:
@siddirp I have not had the chance to go back and try updating the bootloader yet, but I will report back once I try doing so.

chunhochow, 2019-05-02:
@siddirp I think you meant `bootloadUpdateBootloaderEx` not `bootloadUpdateBootloader` on the May 26 comment.

I tried uploading the bootloader using the described method, and got:
```
./bin/ISBootloaderExample /dev/ttyUSB0 ~/Downloads/SAMS70-Bootloader_v2c_r1.5.bin
Bootloader failed! Error: Failed to handshake with SAM-BA
```

siddirp, 2019-05-06:
Make sure this line is in the version of the SDK you have:
https://github.com/inertialsense/InertialSenseSDK/blob/master/src/inertialSenseBootLoader.c#L1433

In older versions of the SDK, that parameter wasn't set correctly.

*Andrew Priddis | Sr Systems Engineer **|* *Inertial Sense, INC.*

Mobile: (801) 610-6359 | InertialSense.com <http://www.inertialsense.com/>

siddirp, 2019-07-18:
Updates from Inertial Sense

http://twitter.com/intent/tweet?text=Inertial+Sense+-+Summer+2019+Update+and+Dual+GNSS-INS+Compass+Product+Launch:&utm_source=Inertial+Sense%2C+Inc.&utm_campaign=84319fe5ac-SummerNewsletter_2019&utm_medium=email&utm_term=0_6b7f39d3ad-84319fe5ac-69182201&mc_cid=84319fe5ac&mc_eid=[UNIQID] https%3A%2F%2Fmailchi.mp%2Finertialsense%2Finertial-sense-quarterly-update-summer2019 Tweet (http://twitter.com/intent/tweet?text=Inertial+Sense+-+Summer+2019+Update+and+Dual+GNSS-INS+Compass+Product+Launch:&utm_source=Inertial+Sense%2C+Inc.&utm_campaign=84319fe5ac-SummerNewsletter_2019&utm_medium=email&utm_term=0_6b7f39d3ad-84319fe5ac-69182201&mc_cid=84319fe5ac&mc_eid=[UNIQID] https%3A%2F%2Fmailchi.mp%2Finertialsense%2Finertial-sense-quarterly-update-summer2019)
http://www.linkedin.com/shareArticle?url=https%3A%2F%2Fmailchi.mp%2Finertialsense%2Finertial-sense-quarterly-update-summer2019&mini=true&title=Inertial+Sense+-+Summer+2019+Update+and+Dual+GNSS-INS+Compass+Product+Launch&utm_source=Inertial+Sense%2C+Inc.&utm_campaign=84319fe5ac-SummerNewsletter_2019&utm_medium=email&utm_term=0_6b7f39d3ad-84319fe5ac-69182201&mc_cid=84319fe5ac&mc_eid=[UNIQID] Share (http://www.linkedin.com/shareArticle?url=https%3A%2F%2Fmailchi.mp%2Finertialsense%2Finertial-sense-quarterly-update-summer2019&mini=true&title=Inertial+Sense+-+Summer+2019+Update+and+Dual+GNSS-INS+Compass+Product+Launch&utm_source=Inertial+Sense%2C+Inc.&utm_campaign=84319fe5ac-SummerNewsletter_2019&utm_medium=email&utm_term=0_6b7f39d3ad-84319fe5ac-69182201&mc_cid=84319fe5ac&mc_eid=[UNIQID])
http://us18.forward-to-friend.com/forward?u=5728d9f91f0bc59010d402ad1&id=84319fe5ac&e=00884d6621 Forward (http://us18.forward-to-friend.com/forward?u=5728d9f91f0bc59010d402ad1&id=84319fe5ac&e=00884d6621)

Newsletter - Summer 2019

Hooray for Summer! This time of year means many of you are out conducting field tests or taking some much deserved time off.  We hope you have a great time out doing both. The Inertial Sense team is happy to announce the general availability of our Dual GNSS-INS Compass product.  This product was produced to primarily solve one thing; the headaches of magnetometers.

In addition to the new product, you'll also find information about our 1.7.6 software update and a fun customer case study.

Enjoy,

-The Inertial Sense Team

** News and Updates
------------------------------------------------------------

** Announcing General Availability of Dual GNSS-INS Compass
------------------------------------------------------------
We are pleased to announce the general availability of a line of dual GNSS heading sensors that provideaccurate heading measurements without need of vehicle dynamics or magnetic sensors. This heading solution, based on dual GNSS on-board receivers, provides a .3 deg heading accuracy using a 1 meter antenna distance baseline.

Learn More (https://inertialsense.com/inertial-sense-releases-dual-gnss-ins-compassing-solution/?utm_source=Inertial+Sense%2C+Inc.&utm_campaign=84319fe5ac-SummerNewsletter_2019&utm_medium=email&utm_term=0_6b7f39d3ad-84319fe5ac-69182201&mc_cid=84319fe5ac&mc_eid=[UNIQID])

** New EVALUATION BOARD w/Logger and Connectivity
------------------------------------------------------------
To support a quick and easy test drive of our products we are releasing the second generation of our Evaluation Board.  The EVB2 comes loaded with features such as onboard logging to microSD card, XBee radio and WiFi/Bluetooth capabilities.

RTK users will love the pre-configured 915 MHz radios so that our units can easily be used in both a Rover and Base station configuration.
Get Started (https://inertialsense.com/product/%c2%b5ins-dual-gnss-compass-development-kit-rugged-evb2/?utm_source=Inertial+Sense%2C+Inc.&utm_campaign=84319fe5ac-SummerNewsletter_2019&utm_medium=email&utm_term=0_6b7f39d3ad-84319fe5ac-69182201&mc_cid=84319fe5ac&mc_eid=[UNIQID])

** New Firmware Features Released - 1.7.6
------------------------------------------------------------
We continue to update our firmware based on your feedback. We're currently on version 1.7.6.  You can always find the latest release as well as product documentation on our resources page.

Key Features
* Official Dual GNSS-INS Compassing for accurate heading measurements without the need for vehicle dynamics or magnetic sensors.
* Open-source project for EVB-2 now available.
* Toggle between RTK positioning mode and Dual GNSS compassing mode without reset.
* Strobe input is now interrupt-based instead of timer-based

Link to Latest Release (https://inertialsense.com/resources/?utm_source=Inertial+Sense%2C+Inc.&utm_campaign=84319fe5ac-SummerNewsletter_2019&utm_medium=email&utm_term=0_6b7f39d3ad-84319fe5ac-69182201&mc_cid=84319fe5ac&mc_eid=[UNIQID])

** Ceres Imaging - Customer Case Study
Aerial imagery company saves money and time with  INS/GNSS sensors

------------------------------------------------------------
Read Case Study (https://inertialsense.com/aerial-imagery-companies-use-drones-ins-gnss?utm_source=newsletter&utm_medium=email&utm_campaign=ceres_case_study&utm_term=sensor&utm_source=Inertial+Sense%2C+Inc.&utm_campaign=84319fe5ac-SummerNewsletter_2019&utm_medium=email&utm_term=0_6b7f39d3ad-84319fe5ac-69182201&mc_cid=84319fe5ac&mc_eid=[UNIQID])

** Industry Info
------------------------------------------------------------
[... cut ...]
 You are receiving this email because you opted in via our website.

Our mailing address is:
Inertial Sense, Inc.
384 S 400 W
Suite 125
Lindon, UT 84042
USA
You received this email because you've provided information to Inertial Sense.
You can ** update your preferences (https://inertialsense.us18.list-manage.com/profile?u=5728d9f91f0bc59010d402ad1&id=6b7f39d3ad&e=00884d6621)
or ** unsubscribe from this list (https://inertialsense.us18.list-manage.com/unsubscribe?u=5728d9f91f0bc59010d402ad1&id=6b7f39d3ad&e=00884d6621&c=84319fe5ac)
.

siddirp, 2019-08-16:
WHY HIGH-PERFORMANCE LIDAR SENSOR MANUFACTURERS PAIR INS/GNSS FOR 3D AERIAL MAPPING

http://www.facebook.com/sharer/sharer.php?u=https%3A%2F%2Fmailchi.mp%2Finertialsense%2Fcustomer-case-study-pairing-gnss-ins-sensor-with-lidar-for-3d-aerial-mapping&utm_source=Inertial+Sense%2C+Inc.&utm_campaign=64f369d0cd-EMAIL_CAMPAIGN_2019_08_15_04_00&utm_medium=email&utm_term=0_6b7f39d3ad-64f369d0cd-69182201&mc_cid=64f369d0cd&mc_eid=[UNIQID] Share (http://www.facebook.com/sharer/sharer.php?u=https%3A%2F%2Fmailchi.mp%2Finertialsense%2Fcustomer-case-study-pairing-gnss-ins-sensor-with-lidar-for-3d-aerial-mapping&utm_source=Inertial+Sense%2C+Inc.&utm_campaign=64f369d0cd-EMAIL_CAMPAIGN_2019_08_15_04_00&utm_medium=email&utm_term=0_6b7f39d3ad-64f369d0cd-69182201&mc_cid=64f369d0cd&mc_eid=[UNIQID])
http://twitter.com/intent/tweet?text=Customer+Case+Study+-+Pairing+GNSS-INS+sensor+with+LiDAR+for+3d+Aerial+Mapping:&utm_source=Inertial+Sense%2C+Inc.&utm_campaign=64f369d0cd-EMAIL_CAMPAIGN_2019_08_15_04_00&utm_medium=email&utm_term=0_6b7f39d3ad-64f369d0cd-69182201&mc_cid=64f369d0cd&mc_eid=[UNIQID] https%3A%2F%2Fmailchi.mp%2Finertialsense%2Fcustomer-case-study-pairing-gnss-ins-sensor-with-lidar-for-3d-aerial-mapping Tweet (http://twitter.com/intent/tweet?text=Customer+Case+Study+-+Pairing+GNSS-INS+sensor+with+LiDAR+for+3d+Aerial+Mapping:&utm_source=Inertial+Sense%2C+Inc.&utm_campaign=64f369d0cd-EMAIL_CAMPAIGN_2019_08_15_04_00&utm_medium=email&utm_term=0_6b7f39d3ad-64f369d0cd-69182201&mc_cid=64f369d0cd&mc_eid=[UNIQID] https%3A%2F%2Fmailchi.mp%2Finertialsense%2Fcustomer-case-study-pairing-gnss-ins-sensor-with-lidar-for-3d-aerial-mapping)
http://us18.forward-to-friend.com/forward?u=5728d9f91f0bc59010d402ad1&id=64f369d0cd&e=00884d6621 Forward (http://us18.forward-to-friend.com/forward?u=5728d9f91f0bc59010d402ad1&id=64f369d0cd&e=00884d6621)

Customer Case Study: Livox LiDAR
Why LiDAR Sensor Manufacturers Pair GNSS-INS for 3D Aerial Mapping

** The story
------------------------------------------------------------
LiDAR sensor manufacturers such as Livox Lidar are pairing Inertial Navigation Sensors (GNSS-INS) to help generate accurate 3D maps for their clients.
Read how Livox integrated the uINS+RTK sensor from Inertial Sense to scan the Huizhou Bay Bridge located in Huizhou, Guangdong, China.
Read Case Study Here (https://inertialsense.com/lidar-sensors-manufacturers-ins-gnss-3d-aerial-mapping/?utm_source=newsletter&utm_medium=email&utm_campaign=Livox&utm_source=Inertial+Sense%2C+Inc.&utm_campaign=64f369d0cd-EMAIL_CAMPAIGN_2019_08_15_04_00&utm_medium=email&utm_term=0_6b7f39d3ad-64f369d0cd-69182201&mc_cid=64f369d0cd&mc_eid=[UNIQID])

** Inertial Sense Sensors
------------------------------------------------------------
Our family of IMU/AHRS/INS and Dual GNSS-INS sensors fit a variety of application needs.

With 0.1 degree roll/pitch accuracies and centimeter-level RTK positioning, we cover a broad set of use cases.

Let us know how we can help.

Browse Sensors (https://inertialsense.com/products/?utm_source=Inertial+Sense%2C+Inc.&utm_campaign=64f369d0cd-EMAIL_CAMPAIGN_2019_08_15_04_00&utm_medium=email&utm_term=0_6b7f39d3ad-64f369d0cd-69182201&mc_cid=64f369d0cd&mc_eid=[UNIQID])

============================================================
** Facebook (http://www.facebook.com?utm_source=Inertial+Sense%2C+Inc.&utm_campaign=64f369d0cd-EMAIL_CAMPAIGN_2019_08_15_04_00&utm_medium=email&utm_term=0_6b7f39d3ad-64f369d0cd-69182201&mc_cid=64f369d0cd&mc_eid=[UNIQID])
** Twitter (http://www.twitter.com/?utm_source=Inertial+Sense%2C+Inc.&utm_campaign=64f369d0cd-EMAIL_CAMPAIGN_2019_08_15_04_00&utm_medium=email&utm_term=0_6b7f39d3ad-64f369d0cd-69182201&mc_cid=64f369d0cd&mc_eid=[UNIQID])
** Link (http://www.instagram.com/?utm_source=Inertial+Sense%2C+Inc.&utm_campaign=64f369d0cd-EMAIL_CAMPAIGN_2019_08_15_04_00&utm_medium=email&utm_term=0_6b7f39d3ad-64f369d0cd-69182201&mc_cid=64f369d0cd&mc_eid=[UNIQID])
** Website (http://mailchimp.com?utm_source=Inertial+Sense%2C+Inc.&utm_campaign=64f369d0cd-EMAIL_CAMPAIGN_2019_08_15_04_00&utm_medium=email&utm_term=0_6b7f39d3ad-64f369d0cd-69182201&mc_cid=64f369d0cd&mc_eid=[UNIQID])
Copyright © 2019 Inertial Sense, Inc., All rights reserved.
 You are receiving this email because you opted in via our website.

Our mailing address is:
Inertial Sense, Inc.
384 S 400 W
Suite 125
Lindon, UT 84042
USA
Want to change how you receive these emails?
You can ** update your preferences (https://inertialsense.us18.list-manage.com/profile?u=5728d9f91f0bc59010d402ad1&id=6b7f39d3ad&e=00884d6621)
or ** unsubscribe from this list (https://inertialsense.us18.list-manage.com/unsubscribe?u=5728d9f91f0bc59010d402ad1&id=6b7f39d3ad&e=00884d6621&c=64f369d0cd)
.

siddirp, 2020-05-14:
@chunhochow I'm closing this ticket for now. Please reopen if needed.

closed 2020-05-14 by siddirp