inertialsense/inertial-sense-ros-old #27: Not getting any output for /gps topic

Dinatorehanova, 2019-06-23:
Hi,
When i am trying to run the command like "rostopic echo /gps" I am getting the following message
[image]

But at the same place I can get the gps data using the eval tool in the windows platform.Please help me to resolve this issue.

Is it because I did not update the firmware in the ros for Uins ? When I am trying to update it I am getting this error 

[image]

Thank you very much for your valuable time. I will wait for your kind reply.

superjax, 2019-06-24:
This is probably a ROS environment error.

Have you sourced the `setup.bash` in your `catkin_ws/devel/setup.bash`?

http://wiki.ros.org/catkin/Tutorials/create_a_workspace

Dinatorehanova, 2019-06-24:
@superjax  I did and all of my  other topics works fine except the /gps topic all my other topics ares working.

I am doing the following

cd myworkspace

source devel/setup.bash

[image]

And this is coming up

superjax, 2019-06-24:
could you paste the results of the following?

`rospack profile`
`rospack find inertial_sense`

If the second command fails, could you paste full path to the `inertial_sense` directory?

For reference, this is the report from my machine:
``` bash
$ rospack profile
0.008310   /opt/ros/melodic/share
0.000957   /home/superjax/ws/src
```

``` bash
$ rospack find inertial_sense
/home/superjax/ws/src/inertial_sense_ros
```

I'm primarily checking that the `inertial_sense_ros` workspace is listed under `rospack profile`

Dinatorehanova, 2019-06-25:
@superjax 
[image]

[image]

ok I am getting an error with the rospack find inertial sense. My all other topics are working except this /gps.

Please help me with it.

Thank you for your time

superjax, 2019-06-25:
The reason that `inertial_sense` is not found is that the `myworkspace/setup/devel.bash` is only sourced in the terminal you use to run the node,  therefore, any other terminals don't have the path in their environment.

Try adding `source `/myworkspace/setup/devel.bash` to the bottom of your `~/.bashrc` file.  The `~/.bashrc` file is a script that runs every time you open a new terminal.

For example, here is mine:

``` bash
$ tail ~/.bashrc -n 3
# ROS_WORKSPACE
source /opt/ros/melodic/setup.bash
source ~/ws/devel/setup.bash
```

(The `tail` command just prints out the bottom "n" lines of a file)

Dinatorehanova, 2019-06-25:
Hi,

Thank you for your kind reply.

If I understand you correcly,

[image]

[image]

Is that ok ?

closed 2019-06-28 by Dinatorehanova