inertialsense/inertial-sense-ros-old #28: /IMU topic Orientation remains same

Dinatorehanova, 2019-06-25:
@superjax , is it normal to have a continuous value of 0 under the /imu topic for orientation? It seems the value of my this topic does not change at all.Even when my vehicle is moving or tilting it always remains 0 for the orientation part while the other data like linear acceleration and angular velocity changes for each time stamp.

Thanks

siddirp, 2019-06-25:
@Dinatorehanova That is normal. The /imu topic will publish the IMU data as linear accel and angular velocity. These values are raw IMU data directly from the onboard IMU. Attitude information can be found in the /ins topic under pose/pose/orientation.

Dinatorehanova, 2019-06-25:
@siddirp , thank you very much for your kind reply. Ok I understand but is it possible that I can retrieve the orientation data in the imu also? If so, can you tell me how I can do that ?

Thanks for your time

siddirp, 2019-06-25:
@Dinatorehanova  Do you mean you want the INS orientation output to be published as part of the /imu topic? That is probably possible, but you'd need to rewrite some of the node code. If you can give more background I can give you better direction. What is the end goal?

Dinatorehanova, 2019-06-25:
@siddirp  yes you are correct. I want that the /imu topic will publish the orientation information also.Right now it only gives the linear acceleration and the angular velocity.As /imu topic I need it to publish the orientation information also.Since the /ins is a (navigation/odometry) topic I can not use it for my purpose.

siddirp, 2019-06-25:
@Dinatorehanova You could rewrite the node to meet your needs. For example, the INS2 callback function could be changed to publish a sensor_msg which seems to be what you need. Or you could track the latest orientation information within the INS2 message and publish it along with the IMU data.

Dinatorehanova, 2019-06-25:
@siddirp , can you please help me to modify the code? 

I am quite newbie to the ros environment.

Dinatorehanova, 2019-06-25:
@siddirp  you mean like here 

[image]

Can you tell me how to write the function ?

siddirp, 2019-06-26:
This works: [image]

Dinatorehanova, 2019-06-26:
@siddirp 

Do I need to uncomment the line 376 as well ?

siddirp, 2019-06-26:
I ran what I showed in the screenshot and it worked well.

Dinatorehanova, 2019-06-26:
@siddirp ,

Thank you for being patient with me, 
I followed the code that you provided but I do not know still no t getting any change with the orientation data under the /imu topic for each time stamp
[image]

[image]

siddirp, 2019-06-26:
Is that second image your code? You didn't add anything to it.... Look at the code in the image I sent. Notice how it takes values from the odom_msg and puts them into the imu1_msg.

Dinatorehanova, 2019-06-26:
@siddirp , sorry for being naive.It works.Thank you very much for your kind co-operation.

Thanks

closed 2019-06-26 by siddirp