inertialsense/inertial-sense-ros-old #34: INS topic clarification(%time and field.header.stamp)

ethan20120, 2019-09-10:
Hi,

Can anyone explain what's the difference between these two columns(%time and field.header.stamp) under the /ins topic ? It seems like both of them represents the timestamp.

Thanks

siddirp, 2019-09-10:
Can you elaborate a little bit? I'm not sure what columns you're referring to....

ethan20120, 2019-09-10:
[image]

Hi @siddirp ,

Thanks for your reply. I have marked the columns. I just converted the ros message of the /ins topic into the csv purpose for the visualization purpose. Looks like both of the columns contains the timestamp. Can you please clarify what's the difference in between ? And does the position and the orientation data is coming in the NED frame by default?

Thanks 
Ethan

superjax, 2019-09-10:
The `time` field appears to be from the metadata of the message (the time the message arrived to whatever you are using to log the data).  Whereas the `header.stamp` is the timestamp, logged at the time of the measurement.

This isn't specific to the `inertial_sense_ros` node.  This a very common practice across ROS in general. It is very standard to add a `header` to messages, which include a time stamp.  If you look really carefully, I suspect there are a few milliseconds difference between `time` and the `field.header.stamp` because it takes some time for the message to actually hit whatever is logging messages.

ethan20120, 2019-09-10:
Hi @superjax 

Thanks for your reply. Just a quick question by default the rotation(quaternion), translation(position) is coming in the NED frame right ?

Thanks

superjax, 2019-09-11:
Yes.  They are both in NED.

I believe there are parameters to change the output frame if you need to as well.

siddirp, 2019-09-11:
Thanks @superjax !

@ethan20120  We have added code to change frames, but that hasn't been officially released in our Github node. When we do our next firmware release the node will be updated. In the meantime changing to ENU frame was described in a previous issue. See: https://github.com/inertialsense/inertial_sense_ros/issues/29

superjax, 2019-09-11:
@siddirp, thanks for the clarificaton.  Sorry about any confusion.

I was mistaking the `INS_xyz` and `INS_rpy` parameters, which are similar, but not quite the same thing.

ethan20120, 2019-09-11:
@siddirp 
Thanks for the reference. So, it seems like there will be option when running he node to select from ENU and NED?

@superjax , I believe the x,y,z position data coming from the INS is translation while the orientation rotation quaternion in the for of x,y,z,w in NED form with W being the real number. Just a quick crosscheck what are the data that are coming in the sensor body frame? The rotation and the position data are in the NED frame.

Thanks.

siddirp, 2019-09-13:
@ethan20120 That is correct. It may be a while before that is released so it might be best to implement yourself for now.
The position data is north, east and down offset from reference latitude, longitude, and altitude to current latitude, longitude, and altitude
Quaternion body rotation with respect to NED: W, X, Y, Z

closed 2019-09-17 by siddirp