inertialsense/inertial-sense-ros-old #35: Magnetometer Calibration

ethan20120, 2019-09-17:
Hi !

Is it necessary to do the sensor calibration (magnetometer ) once in a while to get proper data from the field?

https://docs.inertialsense.com/user-manual/system-config/magnetometer/

We are collecting data using the mobile platform. Should we have to calibrate the sensor ?

Thanks 
Ethan

siddirp, 2019-09-17:
@ethan20120  We usually do a magnetic calibration before a test although after you do it once it is saved onboard. Also, during use the calibration is constantly updated in the background.

ethan20120, 2019-09-17:
Hi @siddirp ,

So do I have to calibrate the sensor in order to get the correct information or it is updating by itself continuously like you said ?

Thanks 
Ethan

siddirp, 2019-09-18:
It's good practice to ensure you have a good calibration. Usually you don't need to re-calibrate after a calibration because the calibration is saved. If the environment in which the device is mounted changes (hard or soft ferrous metals installed nearby) in ways that affect the magnetometer then it would indeed be needed.

closed 2020-05-14 by siddirp