inertialsense/inertial-sense-ros-old #36: Eroor in Cartesian to Geodetic Transformation

ethan20120, 2019-10-06:
Hi,

we are trying to transform the position data coming up from the ins in ENU coordinate system to the geodetic coordinate system so that we can match up the location where the gps is missing. However, we are getting very weird value.For example the gps coordinate of our area is 
41.7458322 , -111.8035665, in lat,long, and altitude. And the coordinate(position x,y,z) of the corresponding place coming from the ins is -11328.84277, 166217.9844,  -106.8024216.  It should match up with the coordinate of the gps after the transformation but it does not. we are using matlab enu2geodetic transformation function.

Does it have anything to do with the ins calibration ?

Any help would be very appreciated.
Thanks

superjax, 2019-10-07:
Have you set your reference lla?

See the 'set_ref_lla' service.

ethan20120, 2019-10-07:
Hi,

Thank you very much for your kind reply. Ok, I got your point. But can you tell me something line the data that i have collected before setting up the ref_lla, will that be garbage  or I can somehow modify the data ?

ethan20120, 2019-10-07:
@superjax , we just changed the reference but it is still giving us error. Does it has anything to do with the calibration ? If so, how to do this calibration ?

siddirp, 2019-10-07:
Hi @ethan20120 ,
Can you give a more in depth example of your input parameters? The [matlab enu2geodetic function](https://www.mathworks.com/help/map/ref/enu2geodetic.html) requires 
xEast, yNorth, zUp, lat0, lon0, h0, and spheroid as input parameters.

This may be obvious but I'm going to describe everything just to make sure you don't miss anything. lat0, lon0, and h0 should match the reference Lat, Lon, and Alt used in the uINS. This represents the point from which the ENU coordinates will be measured.

 You can enter this manually in ROS (set_refLLA_value) or you can set the current location of the uINS as the reference LLA(set_refLLA_current). These values can also be set or obtained using the EvalTool. 

xEasy, yNorth, zUp are the coordinates in meters of the point of which you want to calculate the LLA position. The uINS outputs an NED location in x,y,z coordinates so you need to transform NED to ENU.

spheroid should be wgs84Ellipsoid which is the spheroid that is used by GPS systems.

lat, lon, h (the output) is the location that is calculated. To reiterate, this is the point measured in meters from the reference LLA and converted to lat lon, alt.

Hopefully that helps. If that doesn't clear things up please send more information or data so we can dig deeper.

-Andrew Priddis
Inertial Sense Support

ethan20120, 2019-10-07:
Hi Andrew ,

Please correct us if we are wrong we need to calculate the offset between the gps and ins by ourselves right ?

The following picture represents the vehicle trajectory (white dot from gps and black from the ins)

[image]

Now after doing the offset calculation we get the 

[image]

Still the gps and the ins is not aligning properly. I guess it's because of the drifting error of the ins right ?

ethan20120, 2019-10-08:
Hi @siddirp ,

Our INS and GPS are not aligning. Can you please give us some hints on how to resolve this ?

Thanks 
Ethan

Chenweng1, 2019-10-24:
Hi @siddirp 

we have updated the firmware to the latest version 1.7.6 but still we are facing the same error that the INS and the GPS points are not overlapping with each other. Any help will be very appreciated. Also can you please tell us how we can increase the update rate of the INS. Currently it's updating in every 10 milliseconds. Can we update it faster than this?

Thanks 
Weng

siddirp, 2020-05-14:
@Chenweng1 - Since we discussed this thoroughly over the support email I am closing this issue.

closed 2020-05-14 by siddirp