inertialsense/inertial-sense-ros-old #38: Parameter for receiving RTK corrections from radio serial port?

dominoc, 2019-10-17:
In the EvalTool or the cltool, you can configure the uINS as a rover to receive RTK corrections via radio through a serial port e.g.
cltool.exe -flashConfig=rtkCfgBits=0x01 -svr=RTCM3:SERIAL:COM9:57600
I can't seem to find the same in the ROS driver parameter list. How can I do this?

siddirp, 2019-10-17:
@dominoc - The CLTool documentation can be a little misleading. That -srv function is to stream corrections from a source that is connected using a COM port (e.g. a radio connect a COM port on the computer). Those corrections are captured by the CLTool then passed to a uINS device on the specified COM port. It is basically a pass through to do everything through a computer. 

The uINS can recognize base corrections streamed to any serial port. If the radio has a serial output it can be connected to the serial ports on the uINS directly.

dominoc, 2019-10-18:
Wasn't questioning CLTOOL but I read through the ROS driver source code and figured out the right ROS param values to pass, e.g. RTK_server_ip=SERIAL:COMX and RTK_server_port=9600

closed 2019-10-18 by dominoc

wekorthals, 2020-08-05:
@dominoc I'm trying to do the same thing in ROS on a Raspberry Pi running Ubuntu Mate. Will you please, share more details on how you got it working? I have tried setting my parameters in the launch file as:
RTK_server_ip: "/dev/ttyS0",
RTK_server_port: 57600,
RTK_correction_type: "RTCM3"
/dev/ttyS0 is the serial tx/rx uart port on the RPi and 57600 is the baud rate of the radio streaming RTCM3.
Unfortunatly, this is not working as far as I can tell. I get a message that the RTK server was successfully connected but, the message shows RTCM3:127.0.0.1:57600 and nothing shows up in the topic /RTK/info or rel.

Any help you can provide is much appreciated. Thanks in advance.

dominoc, 2020-08-06:
You need the SERIAL keyword.e.g.
RTK_rover: "true"
RTK_server_ip: "SERIAL:/dev/ttyS0",RTK_server_port: 57600,
RTK_correction_type: "RTCM3"

Best regards
Dominic