inertialsense/inertial-sense-ros-old #45: can't connect to uINS

Ador2, 2020-04-30:
I have changed the baudrate to 19200 and now my uINS does not connect anymore i've tried eval tool cltool every baud there is, when i cennect uINS through usb cable in windows it shows COM number and in linux it sees ttyUSB0 but doesn't connect

celikemir, 2020-11-25:
if you try to connect with ros change port name as uINS port name.

siddirp, 2020-11-25:
Hi @Ador2,
Is it possible that you may have done something that could have done a chip erase? Or, did you just update the firmware? Try clicking the box in the EvalTool to open the COM port, then click update firmware. The uINS has an autobaud feature that is on by default. If you connect and try to communicate at a baud that is not its configured baud rates it will try to match it. You could try to power cycle a few times to see if it will match.

closed 2022-02-23 by siddirp