inertialsense/inertial-sense-ros-old #46: GPS (1) message not published with correct frame information

mjshiggins, 2020-05-14:
The GPS message is assigned the default body frame when it is potentially (and very likely) offset a considerable distance from the INS unit. The ROS driver has a configuration parameter to set this value using the SDK, but it is only used internally on the device for INS / RTK / Compassing functions.

I'd highly suggest not keeping a separate body transform tree, as ROS is equipped to handle static transforms quite well with the URDF spec. It might be a good idea to pull this data from the ROS API (or at least provide that as an option), or alternatively (and not ideally) publish the static transforms from this node using the configured info. At the very least, a separate from should be defined for the GPS antennas.

Additionally, and this is a minor issue, but the parameter in the example launch file `GPS_ant1_xyz` doesn't match what the code is looking for: `GPS_ant1_xyz` or `GPS_ant2_xyz`.

siddirp, 2020-05-14:
@mjshiggins - These are good suggestions. Our current efforts are devoted elsewhere, but we released this project as open source with the intent that users would contribute as new features are recognized. Feel free to modify it as you see fit and create a pull request. Until then, this will remain and issue until we can divert resources back to building out new ROS package features.

mjshiggins, 2020-05-14:
@siddirp Totally understandable. We did this internally, let me see if I can push changes back upstream.

closed 2022-02-23 by siddirp