inertialsense/inertial-sense-ros-old #47: GPS Raw data messages are empty

wekorthals, 2020-06-15:
I have set "stream_GPS_raw: true" in the launch file and "rostopic list" shows the topics are present. However, the topics gps/obs, gps/eph, and gps/geph do not contain any data when echoed to the screen or recorded to a rosbag. All other topics seem to be producing messages as expected.
The rugged uINS is running firmware v1.8.1 and the Raspberry Pi 3 has the most recent inertial_sense_ros wrapper.

siddirp, 2020-06-23:
Are you testing this in a place where your GPS antenna has a clear signal?
I have just tested /gps/eph and /gps/obs and both do indeed work.

[image: Inertial Sense] <http://www.inertialsense.com/>
*Andrew Priddis*
*Sr Systems Engineer*
801.610.6359 (Cell)
andrew.priddis@inertialsense.com
*inertialsense.com* <http://www.inertialsense.com/>

wekorthals, 2020-06-24:
Yes, clear signal. All other messages have data as expected. I can see them as data streams and topics but, the logs are empty. I do have dual compassing turned on.

siddirp, 2020-06-24:
What does mems precision and calibration look like in 2020?

View in browser (https://info.inertialsense.com/e2t/sc2/Mm-nFbk3rnlW4XrjV68zLDvMW7_bP1p53N9G8W5GS7JJ6zKYp8f1HVPN203 )

Inertial Sense (https://info.inertialsense.com/e2t/sc2/Mm-nFbk3rnlW4XrjV68zLDvMW7_bP1p53N9G8W5GS7JJ6zKYp8f1HVPN213 )

The Tiny Sensor with Giant Capabilities

What could your device do with this kind of power?

IS Sensor Datasheet (https://info.inertialsense.com/e2t/sc2/Mm-nFbk3rnlW4XrjV68zLDvMW7_bP1p53N9G8W5GS7JJ6zKYp8f1HVPN223 )

IMU Datasheet

Your device deserves high precision and quick telemetry without being weighed down. Check out our Datasheet and see if our sensors meet your specifications.

Download Datasheet (https://info.inertialsense.com/e2t/sc2/Mm-nFbk3rnlW4XrjV68zLDvMW7_bP1p53N9G8W5GS7JJ6zKYp8f1HVPN233 )

"Inertial Sense has developed a disruptive technology. It is going to change everything about our industry."

- AUVSI Judges, 2019

View Products (https://info.inertialsense.com/e2t/sc2/Mm-nFbk3rnlW4XrjV68zLDvMW7_bP1p53N9G8W5GS7JJ6zKYp8f1HVPN243 )

Schedule Meeting (https://info.inertialsense.com/e2t/sc2/Mm-nFbk3rnlW4XrjV68zLDvMW7_bP1p53N9G8W5GS7JJ6zKYp8f1HVPN253 )

Read Blog Posts (https://info.inertialsense.com/e2t/sc2/Mm-nFbk3rnlW4XrjV68zLDvMW7_bP1p53N9G8W5GS7JJ6zKYp8f1HVPN263 )

Inertial Sense, 384 S 400 W #125th, Lindon, UT 84042, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/manage-preferences/unsubscribe-all?d=VnBg1c6WzZmnW2p76LK3-1rCSW3X_fKX21k003W4rCVKM4tCK7cW3R40yq4tDT0qW41_3pm3KbsvFW3R0gs53M4TzGW47szjy45RlhdW3P28X24hCPvhW43T2P-7DvMj2W17f7hv63pVqzW4XZGPb4WpGqlW53p_fj70cZxBV28YFB7j6w6TMGKd_P20N7wV_ZKN18yD6J7N7w_Xm47-NBm51nQvr1&v=3&_hsenc=p2ANqtz-9wnEnKBvmsVix6Qg2cp4zWKzqGyTsKTo5g6S5EmxW_4a0iv9ALMq4J3eWDuKd09N86qLdqiS9ZPKsAoCWmQtwz_NxirY6nRUsRZV4YqpIemzZisiY8ej4sPZxU3l_VRZl_kjqT9oqeYV2kKwnJEA-2YQBCEQ&_hsmi=86142041 )

Manage preferences (https://info.inertialsense.com/hs/manage-preferences/unsubscribe?d=VnBg1c6WzZmnW2p76LK3-1rCSW3X_fKX21k003W4rCVKM4tCK7cW3R40yq4tDT0qW41_3pm3KbsvFW3R0gs53M4TzGW47szjy45RlhdW3P28X24hCPvhW43T2P-7DvMj2W17f7hv63pVqzW4XZGPb4WpGqlW53p_fj70cZxBV28YFB7j6w6TMGKd_P20N7wV_ZKN18yD6J7N7w_Xm47-NBm51nQvr1&v=3&_hsenc=p2ANqtz-9wnEnKBvmsVix6Qg2cp4zWKzqGyTsKTo5g6S5EmxW_4a0iv9ALMq4J3eWDuKd09N86qLdqiS9ZPKsAoCWmQtwz_NxirY6nRUsRZV4YqpIemzZisiY8ej4sPZxU3l_VRZl_kjqT9oqeYV2kKwnJEA-2YQBCEQ&_hsmi=86142041 )

wekorthals, 2020-06-25:
I ran another test this morning and I am still getting the same results.  I am attaching the text files from running "rostopic hz /gps" as well as /gps/obs, /gps/eph, and /gps/geph. As you can see, the raw topics are being published but, there are not any accompanying messages. The /gps topic does have messages as do all the other ins, imu, etc.

Another issue I had this morning had to do with running "rosservice call /set_refLLA_current" which has worked in the past but, did not work this time. It returned "True" but, the LLA was all set to "0.00".  

[gps_eph_hz.txt](https://github.com/inertialsense/inertial_sense_ros/files/4832766/gps_eph_hz.txt)
[gps_geph_hz.txt](https://github.com/inertialsense/inertial_sense_ros/files/4832767/gps_geph_hz.txt)
[gps_hz.txt](https://github.com/inertialsense/inertial_sense_ros/files/4832768/gps_hz.txt)
[gps_obs_hz.txt](https://github.com/inertialsense/inertial_sense_ros/files/4832770/gps_obs_hz.txt)

/gps is publishing messages.
[gps_msg.txt](https://github.com/inertialsense/inertial_sense_ros/files/4832769/gps_msg.txt)

siddirp, 2020-07-09:
Hi Bill,
Can you send an email to support@inertialsense.com with the details of this
issue?

[image: Inertial Sense] <http://www.inertialsense.com/>
*Andrew Priddis*
*Sr Systems Engineer*
801.610.6359 (Cell)
andrew.priddis@inertialsense.com
*inertialsense.com* <http://www.inertialsense.com/>

siddirp, 2020-07-10:
What does mems precision and calibration look like in 2020?

View in browser (https://info.inertialsense.com/e2t/sc2/Mm-nFbjYfZzW89vQVR5Lb_0xW6T6Sv791lTkZW5r_TFv6WzTN5f1HVPN203 )

Inertial Sense (https://info.inertialsense.com/e2t/sc2/Mm-nFbjYfZzW89vQVR5Lb_0xW6T6Sv791lTkZW5r_TFv6WzTN5f1HVPN213 )

The Tiny Sensor with Giant Capabilities

What could your device do with this kind of power?

IS Sensor Datasheet (https://info.inertialsense.com/e2t/sc2/Mm-nFbjYfZzW89vQVR5Lb_0xW6T6Sv791lTkZW5r_TFv6WzTN5f1HVPN223 )

IMU Datasheet

Your device deserves high precision and quick telemetry without being weighed down. Check out our Datasheet and see if our sensors meet your specifications.

Download Datasheet (https://info.inertialsense.com/e2t/sc2/Mm-nFbjYfZzW89vQVR5Lb_0xW6T6Sv791lTkZW5r_TFv6WzTN5f1HVPN233 )

"Inertial Sense has developed a disruptive technology. It is going to change everything about our industry."

- AUVSI Judges, 2019

View Products (https://info.inertialsense.com/e2t/sc2/Mm-nFbjYfZzW89vQVR5Lb_0xW6T6Sv791lTkZW5r_TFv6WzTN5f1HVPN243 )

Schedule Meeting (https://info.inertialsense.com/e2t/sc2/Mm-nFbjYfZzW89vQVR5Lb_0xW6T6Sv791lTkZW5r_TFv6WzTN5f1HVPN253 )

Read Blog Posts (https://info.inertialsense.com/e2t/sc2/Mm-nFbjYfZzW89vQVR5Lb_0xW6T6Sv791lTkZW5r_TFv6WzTN5f1HVPN263 )

Inertial Sense, 384 S 400 W #125th, Lindon, UT 84042, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/manage-preferences/unsubscribe-all?d=VnBg1c6WzZmnW2p76LK3-1rCSW3X_fKX21hMH5W3T2--243RY-MW41YsTs3Y1Gz9W3DMjPL3KbsvFW3R0gs53M4TzGW47szjy45RlhdW3P28X24hCPvhW43T2P-7DvMj2W17f7hv5rpct5W75VsZF6G5VzGW5w6hPz9lH58kV28ML27LWtN-MGKd_P20N4MW475kmh31s8RYW34VMj58v4Svd4l-VB1&v=3&_hsenc=p2ANqtz---AahyJFrbH5E6B7j6g3cFbrB7Q0j3N8tFHcniv0zdpaPT6t5RUXkyQc2gBmZRteH1BvKHpWxhJpbLUA1O4qu6gv24YalDjMJYSnO76kalaEsmrch-kUgIs0WdiMpIW3FWngW5LQdCtkiHqQR0IDsx4E2WeQ&_hsmi=86142041 )

Manage preferences (https://info.inertialsense.com/hs/manage-preferences/unsubscribe?d=VnBg1c6WzZmnW2p76LK3-1rCSW3X_fKX21hMH5W3T2--243RY-MW41YsTs3Y1Gz9W3DMjPL3KbsvFW3R0gs53M4TzGW47szjy45RlhdW3P28X24hCPvhW43T2P-7DvMj2W17f7hv5rpct5W75VsZF6G5VzGW5w6hPz9lH58kV28ML27LWtN-MGKd_P20N4MW475kmh31s8RYW34VMj58v4Svd4l-VB1&v=3&_hsenc=p2ANqtz---AahyJFrbH5E6B7j6g3cFbrB7Q0j3N8tFHcniv0zdpaPT6t5RUXkyQc2gBmZRteH1BvKHpWxhJpbLUA1O4qu6gv24YalDjMJYSnO76kalaEsmrch-kUgIs0WdiMpIW3FWngW5LQdCtkiHqQR0IDsx4E2WeQ&_hsmi=86142041 )

siddirp, 2020-07-14:
What does mems precision and calibration look like in 2020?

Inertial Sense (https://info.inertialsense.com/e2t/sc2/MmZ-8ykF_TzW64WbrZ4St3bCW5FTy8S7F59FhW477_Rx6PVHlbdBzP8M04 )

The Guide to Choosing A Sensor

Which sensor make sense for your project?

Sensor Buying Cheat Sheet (https://info.inertialsense.com/e2t/sc2/MmZ-8ykF_TzW64WbrZ4St3bCW5FTy8S7F59FhW477_Rx6PVHlbdBzP8N04 )

Sensor Buying Cheat Sheet

At this point, you might be wondering where to start when shopping for a sensor for your project. Deciding between an IMU, AHRS, or GNNS & RTK can be difficult. We've made this handy guide to help answer some basic questions quickly.

Read Blog Post (https://info.inertialsense.com/e2t/sc2/MmZ-8ykF_TzW64WbrZ4St3bCW5FTy8S7F59FhW477_Rx6PVHlbdBzP8P04 )

"Inertial Sense has developed a disruptive technology. It is going to change everything about our industry."

- AUVSI Judges, 2019

View Products (https://info.inertialsense.com/e2t/sc2/MmZ-8ykF_TzW64WbrZ4St3bCW5FTy8S7F59FhW477_Rx6PVHlbdBzP8Q04 )

Schedule Meeting (https://info.inertialsense.com/e2t/sc2/MmZ-8ykF_TzW64WbrZ4St3bCW5FTy8S7F59FhW477_Rx6PVHlbdBzP8R04 )

Read Blog Posts (https://info.inertialsense.com/e2t/sc2/MmZ-8ykF_TzW64WbrZ4St3bCW5FTy8S7F59FhW477_Rx6PVHlbdBzP8S04 )

Inertial Sense, 384 S 400 W #125th, Lindon, UT 84042, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/manage-preferences/unsubscribe-all?d=VnzSyB6WzZmnW2p76LK3-1rCSW3X_fKX21hMH5W3T2--243RY-MW41YsTs3Y1Gz9W3DMjPL3KbsvFW3R0gs53M4TzGW47szjy45RlhdW3P28X24hCPvhW43T2P-7DvMj2W15gBJz8S2pM2W6srpRC4RsznPW4NWzMs81Z5YtVWF3X78RWGQnMXk_7Qbw6lZW4gccJm7WZ1rGW5jZ86j7jY2F-3pBN2&v=3&_hsenc=p2ANqtz-9rH4vhT2btKeF9lLVsdUWl_bk_HPQ2s9vsWC2GDyJm57tY8RAtmfikgYl5Ke_6iEaKLQ4zQ8QR3PJVcInLHX3kXVb1zX4Tqqo07YB7SCggHLaYnAvMzm_NL-SeMdgS4txsuJ39f7JeEvoL1RVuIOEWT8Cr6g&_hsmi=86142139 )

Manage preferences (https://info.inertialsense.com/hs/manage-preferences/unsubscribe?d=VnzSyB6WzZmnW2p76LK3-1rCSW3X_fKX21hMH5W3T2--243RY-MW41YsTs3Y1Gz9W3DMjPL3KbsvFW3R0gs53M4TzGW47szjy45RlhdW3P28X24hCPvhW43T2P-7DvMj2W15gBJz8S2pM2W6srpRC4RsznPW4NWzMs81Z5YtVWF3X78RWGQnMXk_7Qbw6lZW4gccJm7WZ1rGW5jZ86j7jY2F-3pBN2&v=3&_hsenc=p2ANqtz-9rH4vhT2btKeF9lLVsdUWl_bk_HPQ2s9vsWC2GDyJm57tY8RAtmfikgYl5Ke_6iEaKLQ4zQ8QR3PJVcInLHX3kXVb1zX4Tqqo07YB7SCggHLaYnAvMzm_NL-SeMdgS4txsuJ39f7JeEvoL1RVuIOEWT8Cr6g&_hsmi=86142139 )

wekorthals, 2020-07-16:
Update: I rolled back the firmware on the uINS to 1.8.0 and was able to record and view raw GPS messages. Seems the issue is related to name changes in version 1.8.1.

closed 2022-02-23 by siddirp