inertialsense/inertial-sense-ros-old #53: Qu

celikemir, 2020-11-25:
Hi @siddirp 

I try to use uINS with ROS I updated the device v1.8.2. After a successful connection, I try to get quarnation and Roll Pitch Yaw values to get them I used `rostopic echo /imu` but it only returns acc x,y,z and gyro x,y,z.

siddirp, 2020-11-25:
@celikemir,
You'll want to make sure /ins message is enabled and subscribe to that message instead of imu messages.

celikemir, 2020-11-26:
@siddirp 
For sure, after run `rosrun inertial_sense inertial_sense_node` /ins and /imu topics seem to enabled. But `rostopic echo /ins` returns nothing on the other hand `rostopic echo /imu` returns  below data

```
 seq: 942
  stamp: 
    secs: 1606400183
    nsecs: 763668299
  frame_id: "imu"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.00459002144635
  y: 0.024764213711
  z: 0.0142630916089
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: -5.51469612122
  y: 0.200182154775
  z: -8.04611778259
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]

```

Which is include only 6-DOF information. I also test it with test.launch after modification `stream_INS: true,` still same issue there is no ins data.

siddirp, 2020-11-30:
There is code in place to prevent the INS output if there isn't GPS time of
week data. This is in the INS2_callback function. You can edit this as
needed. You will just need to be aware that there is no GPS aid to the
system.

It should be around line 366:

void InertialSenseROS::INS2_callback(const ins_2_t * const msg)
{
  if (!(msg->hdwStatus&HDW_STATUS_GPS_TIME_OF_WEEK_VALID))
  { // Don't run if msg->timeOfWeek is not valid
    return;
  }

[image: Inertial Sense] <http://www.inertialsense.com/>
*Andrew Priddis*
*Sr Systems Engineer*
andrew.priddis@inertialsense.com
*inertialsense.com* <http://www.inertialsense.com/>

closed 2022-02-23 by siddirp