inertialsense/inertial-sense-ros-old #66: Problem when running installation line: git submodule update --init --recursive

maagesen, 2021-05-16:
Hi, 
In the Installation steps for installing the inertial-sense-ros wrapper I am having a problem when running the command: git submodule update --init --recursive
Some of the errors I get are:
-git@github.com: Permission denied (publickey).
-fatal: Could not read from remote repository.
-Failed to clone 'lib/inertial-sense-sdk'. Retry scheduled
-Please make sure you have the correct access rights and the repository exists.
-fatal: clone of 'git@github.com:inertialsense/inertial-sense-sdk.git' into submodule path...

I am very new to github, so my question is: am I doing something wrong, or is it at your end that something has gone wrong?

superjax, 2021-05-18:
That error message probably means the repo is configured to clone via ssh instead of https.

Try setting up git over ssh https://www.freecodecamp.org/news/git-ssh-how-to/ and see it it works.

maagesen, 2021-05-18:
Thank you, it worked perfectly.

closed 2021-05-18 by maagesen

waltjohnson, 2021-05-18:
Thanks, James!

>