inertialsense/inertial-sense-ros-old #96: ERROR: cannot launch node of type [inertial_sense/inertial_sense]: inertial_sense

LauraYan, 2022-02-17:
Hey guys,
 I fetched the code and compiled as your instructions, and then sourced ~/../devel/setup.bash.
 I can use rosrun to open it,  however, as I call "roslaunch inertial_sense_ros test.launch", it gave me this error.
Maybe this is not a really issue, and I may forget something.

Could you give me some suggestion? Thank you in advance.

siddirp, 2022-02-17:
Did you forget to attach there error?

LauraYan, 2022-02-17:
the reported error is 
ERROR: cannot launch node of type [**inertial_sense/inertial_sense_node**]: inertial_sense

 I figured it out.

test.launch 
  <node name="**inertial_sense**_node" pkg="i**nertial_sense**" type="inertial_sense_node"

the project name is "**inertial_sense_ros**", so it should modify like this <node name="inertial_sense_ros_node" pkg="inertial_sense_ros" type="inertial_sense_ros_node" output="screen"/>

LauraYan, 2022-02-17:
i modified like this
 <node name="**inertial_sense_node**" pkg="i**nertial_sense_ros**" type="**inertial_sense_ros_node**" output="screen"/>

LauraYan, 2022-02-17:
firstly
add_executable(**inertial_sense_nod**e src/inertial_sense_node.cpp)
target_link_libraries(inertial_sense_node inertial_sense_ros ${catkin_LIBRARIES})
**inertial_sense_nod into inertial_sense_ros_node**

secondly in test.launch
test.launch
<node name="inertial_sense_node" pkg="inertial_sense" type="inertial_sense_node"

LauraYan, 2022-02-17:
test.launch
<node name="inertial_sense_node" pkg="inertial_sense" type="inertial_sense_node"

modified 
<node name="inertial_sense_ros_node" pkg="inertial_sense_ros" type="inertial_sense_ros_node"

siddirp, 2022-02-23:
Would you like to create a pull request for this change?

LauraYan, 2022-02-24:
Sorry,  I am a newbie of  github, and I  need to learn how to create a pull request.