inertialsense/inertial-sense-ros-old #97: Mag recalibration via rosservice

Srijal97, 2022-02-22:
Hello!

I have a uINS rugged module mounted on my ground vehicle ([like so](https://i.postimg.cc/2jThrq3J/PXL-20220222-010554408-2.jpg)) and I am having trouble getting the absolute orientation from it. When I fuse the yaw output from /ins to my robot_localization node, it causes the robot to rotate on the spot even when it's stationary; and when I use the [imu_madgwick_filter](http://wiki.ros.org/imu_filter_madgwick) package to fuse the /imu and /mag data to get orientation, the orientation output is correct, but when the robot is rotated, the heading immediately returns to the starting position exactly as described [here](https://github.com/ccny-ros-pkg/imu_tools/issues/112). 

I suspect this has a lot to do with the mounting of uINS on the robot - very close to battery cells and the aluminum frame on the bottom. But I am hoping that a mag recalibration can fix this. Also note that the magnetometer readings on /mag seem unreasonably high (like 1.4 Teslas!):
```
magnetic_field: 
  x: -0.695064246655
  y: 1.44425451756
  z: 1.13877093792
magnetic_field_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```
Is this wierd? 

I tried running the `rosservice call /single_axis_mag_cal`, but it does not inform me if the procedure is completed or if I did it correctly? There is only:
```
success: True
message: "Successfully initiated mag recalibration."
```
But how do I know the end of it? 

Please let me know your thoughts. Thanks!

siddharthtelang, 2022-02-24:
I'm also having the same issue.
Is there a fix for this?

siddirp, 2022-02-25:
The magnetometer is very sensitive as it is trying to sense the relatively weak magnetic field of the earth. From the small image you sent, I don't know the amount of current traveling from those batteries, but current in a wire is going to create a magnetic field. As the current fluctuate, the magnetic field does as well. It will be impossible for the magnetometer to decipher the difference. The fix is to carefully design and evaluate the affects of the difference parts of your system on the magnetic field near the uINS.

siddirp, 2022-02-28:
Could you send me a PPD log in .dat format and we can investigate this more closely. 
https://docs.inertialsense.com/user-manual/logging-plotting/data_logging/#logging-ppd-in-cltool

Srijal97, 2022-03-11:
I'm sorry for the delay. Could you please clarify in what condition of the uINS do you want me to record the log? Do you want me to keep it mounted on my robot, or maybe in mid-air? stationary or performing some motion?

siddirp, 2022-03-11:
@Srijal97  Send me a log of you trying to recalibrate the magnetometer. I'd like to see it in your intended integration position.